alaudet/raspi-sump #78: Restyle Web Page

revendoc, 2023-07-16:
Update webpage with new navigation menu using bootstrap.

alaudet, 2023-07-17:
Yup just loaded it up and I like it.  Looks much better on mobile.   I can work with this and will merge this into devel.  Still a couple of tweaks to do but that menu on browser that changes to hamburger menu on smaller screens is much nicer.

* Couple tweaks I will look at are the colour of the blue donate button
* spacing between Water Level Monitoring System and graph
* Version at bottom of screen needs to be adjusted

But awesome, I appreciate this contribution and is a nicer approach.

fitig, 2023-07-18:
Out of curiosity, does the new page "degrade" gracefully when there is no internet?  I could imagine a situation where the pi is powered on but internet is down during a storm and I am nervously / constantly checking my water level.  I suppose I'd probably be opening the pit and just looking myself or just look at the .csv file but I'm lazy :-) .  This is by no means a necessity, just a curiosity.

alaudet, 2023-07-18:
> Out of curiosity, does the new page "degrade" gracefully when there is no internet? I could imagine a situation where the pi is powered on but internet is down during a storm and I am nervously / [...]

As long as there is power to the pi you should be fine.  There is no dependancy on internet access.

revendoc, 2023-07-18:
@alaudet - I just made a few adjustments as a followup to your comments on the last contribution. 

- adjusted the footer to more closely match the original and fixed the layout of the version information
- adjusted the donate button to more closely match the look of linuxnorth.org/donate 
- adjusted some of the padding within the page

alaudet, 2023-07-20:
Thanks changes merged.  Much appreciated

alaudet, 2023-07-21:
It's already looking better and I am good with this for now.  Thanks for your help, much appreciated.

closed 2023-07-21 by alaudet