alaudet/raspi-sump #81: Problem starting systemd after upgrade

nickw56, 2023-07-30:
Failed to reload daemon: Process org.freedesktop.systemd1 exited with status 1
pi@raspberrypi:~ $ systemctl --user start raspisump
Failed to start raspisump.service: Process org.freedesktop.systemd1 exited with status 1
See user logs and 'systemctl --user status raspisump.service' for details.
pi@raspberrypi:~ $ systemctl --user restart raspisump
Failed to restart raspisump.service: Process org.freedesktop.systemd1 exited with status 1
See user logs and 'systemctl --user status raspisump.service' for details.

nickw56, 2023-07-30:
I get the above errors when trying to start the upgraded raspi-sump. Any help gratefully received

nickw56, 2023-07-30:
did a reboot and it now seems to be working ok

nickw56, 2023-07-30:
except the chart is not working! here's the status message I am getting

pi@raspberrypi:~ $ systemctl --user status rsumpwebchart.timer
● rsumpwebchart.timer - RaspiSump Webchart Timer
   Loaded: loaded (/home/pi/.config/systemd/user/rsumpwebchart.timer; enabled; vendor preset: enabled
   Active: inactive (dead)
  Trigger: n/a

alaudet, 2023-07-31:
are you able to run rsumpwebchart.py manually? Do you get errors?

Just type rsumpwebchart.py then check /home/pi/raspi-sump/charts for today.png

Also check /home/pi/raspi-sump/charts/2023/07 to see if it has charts in it.

nickw56, 2023-07-31:
Several reboots later the chart is now working but there's another problem.  The measurement interval is set in the conf file to 300 seconds (5 mins) but what I am seeing in the CSV file does not reflect this: the interval seems to vary between a few seconds up to a few minutes

20:56:45,5.9
20:56:54,5.6
20:57:01,6.2
20:57:08,6.2
20:57:16,6.3
20:57:24,6.1
20:57:32,6.7
20:57:40,6.7
20:57:47,6.8
20:57:55,6.5
20:58:02,6.8
20:58:09,7.1
20:58:16,7.2
20:58:23,7.1
20:58:30,6.9
20:58:36,7.1
20:58:44,7.4
20:58:51,7.6
20:58:58,7.4
20:59:05,7.6

nickw56, 2023-07-31:
Hi Al, I am running 1.9.2 on Bullseye. I got the chart working last night but nothing was running this morning.  Does raspisump start automatically on a re-boot ? If not how can I make this happen? Also still having a problem with erratic measurement intervals.

alaudet, 2023-07-31:
Are your crontab entries still active?

what does `crontab -l` return?

Also please provide the output of 

`systemctl --user status raspisump`

and

`systemctl --user status rsumpwebchart.timer`

alaudet, 2023-07-31:
it still sounds to me like you have remnants of your old install in crontab mixed in with systemd.
Can you post the output of the commands I gave you in the previous message?

alaudet, 2023-07-31:
When it generates a reading every few seconds it indicates that rsump.py is starting, stopping and then systemd tries to restart the service.  (over and over)

Are issues still present?

You wouldnt happen to have an old entry on /etc/rc.local that initiates rsump.py on startup of the pi would you?

Also the commands to make sure raspisump and rsumpwebchart.timer work on reboot are the following

`systemctl --user enable raspisump`
`systemctl --user enable rsumpwebchart.timer`

But it looks to me like they are already enabled from your output.

nickw56, 2023-07-31:
Hi Al,

Nothing in rc.local - see below

# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"

Its been running fine all day, but problem is that rusump.py is running 4 or 5 times a minute when in the config file its set to runn every 300 secs - not a huge issue but a bit puzzling.  Also noticed that I have raspisump.conf and a raspisump.conf.save file - is this intentional? I assume I could delete the .save version?

alaudet, 2023-07-31:
ok

can we just recap quickly

What is working?
What is not working?

nickw56, 2023-07-31:
Hi Al

Sump and chart.timer are both working fine. The remaining issue is the
measurement interval. That is set to 300 secs in config but actually
running every 15 seconds or so.

Thanks for your help

Best regards

*Nick*
*m: 07771 994175*

alaudet, 2023-07-31:
ok let me test this on my end.

alaudet, 2023-07-31:
In the meantime, can you upgrade to my latest development snapshot?  I have included a troublshooting utility to better help me analyse the environment.

This will upgrade you to version 1.9.3rc1

To upgrade run the following

    sudo pip3 install -U --no-binary :all: https://github.com/alaudet/raspi-sump/archive/refs/heads/devel.zip

Then run the following command

    rsumpsupport

This will create a file called `support.txt` in the `/home/pi/raspi-sump/support` folder.

Email me that file at   `alaudet@linuxnorth.org`

You can actually view the contents of that file, there is  no personal info in it.  It's currently a work in progress for my next release but could be helpful here.

nickw56, 2023-07-31:
OK I will do that upgrade and send you the file should I shut down
raspisump and the chart timer first?

Best regards

*Nick*
*m: 07771 994175*

alaudet, 2023-07-31:
yes please shutdown the raspisump and rsumpwebchart.timer, upgrade then start again.

nickw56, 2023-07-31:
Hi Al

Upgrade done.Here's the content of the support.txt file.

Date file generated: 2023-07-31 21:18:55
os version: 10
raspi-sump version: Version 1.9.3rc1
python version on system: Python 3.7.3

Systemctl status for raspisump:
● raspisump.service - RaspiSump Service
   Loaded: loaded (/home/pi/.config/systemd/user/raspisump.service;
enabled; vendor preset: enabled)
   Active: active (running) since Mon 2023-07-31 21:18:50 BST; 5s ago
 Main PID: 24720 (rsump.py)
   CGroup: ***@***.***/raspisump.service
           └─24720 /usr/bin/python3 /usr/local/bin/rsump.py

Jul 31 21:18:50 raspberrypi systemd[692]: Started RaspiSump Service.

Systemctl status for rsumpwebchart.timer:
● rsumpwebchart.timer - RaspiSump Webchart Timer
   Loaded: loaded (/home/pi/.config/systemd/user/rsumpwebchart.timer;
enabled; vendor preset: enabled)
   Active: active (waiting) since Mon 2023-07-31 21:17:47 BST; 1min 8s ago
  Trigger: Mon 2023-07-31 21:30:00 BST; 11min left

Jul 31 21:17:47 raspberrypi systemd[692]: Started RaspiSump Webchart Timer.

Raspi-Sump Variables:
critical_water_level: 10
pit_depth: 35
reading_interval: 300
temperature: 15
trig_pin: 17
echo_pin: 27
unit: metric
smtp_tls: 1
smtp_ssl: 0
alert_when: high
alert_interval: 5
heartbeat: 1
heartbeat_interval: 10080
line_color: FB921D

User Crontab:
# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').
#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command
#*/5 * * * * /usr/local/bin/rsump.py
[... 339 lines cut ...]
2023-07-31 21:09:53,ERROR - Negative reading adjusted to 0.0. Possible
degrading sensor.

Logs/info_log content:
Error executing command: ['cat', '/home/pi/raspi-sump/logs/info_log'].
Return code: 1. Output:

Last 10 lines of logs/heartbeat.log:
2022-04-02 01:15:08,Heartbeat Email Sent
2022-04-09 01:20:09,Heartbeat Email Sent
2022-04-16 01:25:08,Heartbeat Email Sent
2022-04-23 01:30:09,Heartbeat Email Sent
2022-04-30 01:35:08,Heartbeat Email Sent
2022-05-07 01:40:09,Heartbeat Email Sent
2022-05-14 01:45:08,Heartbeat Email Sent
2022-05-21 01:50:09,Heartbeat Email Sent
2022-05-28 01:55:09,Heartbeat Email Sent
2022-06-04 02:00:09,Heartbeat Email Sent

Last 10 lines of logs/alert.log:

Best regards

*Nick*
*m: 07771 994175*

alaudet, 2023-07-31:
Well for one thing you definitely seem to have a degrading sensor as you have multiple 0.0 readings in your error_log.  I am starting to expect that the program is exiting sometimes due to the sensor and systemd tries to restart it right away.  But that doesnt make sense for negative values which are adjusted to 0.0.   You wouldnt happen to have a new sensor laying around?  This one is definitely looking like it is starting to fail.

If that is the issue then I have an issue in the code as well that is not handling this right.

nickw56, 2023-07-31:
Hi AlI will order a new sensor and try that but I doubt that’s the problem as it’s been working ok until now. In the meantime, thanks for all your excellent support.Best RegardsNick

alaudet, 2023-07-31:
ok Nick, I will leave the issue open until then.  In the meantime you could revert to crontab for handling the readings

Just stop and disable the services and change your crontab back.  As well change your reading_interval back to 0 in your raspisump.conf  file.

Al

alaudet, 2023-08-24:
Hi Nick checking in.  We're you ever able to get it going properly?

nickw56, 2023-08-24:
Hi Al, I reverted to running both processes using a cron task and it seems fine.  I think I have a water leak somewhere in or near my property as the sump pump is currently going 5 or 6 times per hour meaning that I am pumping out about 60 - 75 litres per hour and have been for many weeks.  So I am not going to do any more tinkering with the monitoring system until I find the source. The guys at the utility company were impressed when I told them about raspi-sump.Best RegardsNick

alaudet, 2023-08-24:
Thats probably a good idea. I will close this issue and if you need help at some later time we can revisit.
thanks.

closed 2023-08-24 by alaudet