alaudet/raspi-sump #82: rsump.py won't run after upgrade

fitig, 2023-08-03:
After upgrading to 1.9.3 from a pre 1.8 version (with every thing running without issue) when I try to run rsump.py:

```
$ sudo /usr/local/bin/rsump.py
Traceback (most recent call last):
  File "/usr/local/bin/rsump.py", line 12, in <module>
    from raspisump import reading, log, config_values
  File "/usr/local/lib/python3.9/dist-packages/raspisump/reading.py", line 11, in <module>
    from raspisump import log, alerts, heartbeat, config_values
  File "/usr/local/lib/python3.9/dist-packages/raspisump/alerts.py", line 22, in <module>
    configs = config_values.configuration()
  File "/usr/local/lib/python3.9/dist-packages/raspisump/config_values.py", line 66, in configuration
    configs["line_color"] = config.get("charts", "line_color")
  File "/usr/lib/python3.9/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.9/configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'charts'
```

Any ideas what might be going on?

alaudet, 2023-08-03:
Are you sure you are running version 1.9.3?  That nosection error was fixed in version 1.9.2

also you should not be running rsump.py from sudo, but from the regular account.

Check your version number in /home/pi/raspi-sump/VERSION

If you still have errors and are running version 1.9.3, run the command  `rsumpsupport` and post the output of the text file that is created.

fitig, 2023-08-10:
It looks like the VERSION file doesn't exist.  Via a `pip list` I can see that I'm running 1.9.3.  Running rsumpsupport returns this:

```
Traceback (most recent call last):
  File "/usr/local/bin/rsumpsupport", line 71, in <module>
    raspi_sump_version = get_raspi_sump_version(actual_user)
  File "/usr/local/bin/rsumpsupport", line 28, in get_raspi_sump_version
    with open(f"/home/{actual_user}/raspi-sump/VERSION", "r") as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/<removed>/raspi-sump/VERSION'
```

I'm not against a complete re-install if that's the easiest way.

alaudet, 2023-08-11:
ok, looks to me like you did not run the proper install command.  You have to run it exactly as follows

`sudo pip3 install -U --no-binary :all: raspisump`

fitig, 2023-08-11:
Yep that's the command I ran previously.  I ran it again to be sure and it gives the following output:

```
Requirement already satisfied: raspisump in /usr/local/lib/python3.9/dist-packages (1.9.3)
Requirement already satisfied: hcsr04sensor>=1.7 in /usr/local/lib/python3.9/dist-packages (from raspisump) (1.7)
```

alaudet, 2023-08-12:
ok so do the following;

`sudo pip3 uninstall raspisump`

select y to accept

and then 

`sudo pip3 install --no-binary :all: raspisump`

fitig, 2023-08-12:
Ok so I did all of this.  Before I did, I renamed my /home/<removed>/raspi-sump directory to /home/<removed>/raspi-sump.old.  Based on the install instructions here: https://github.com/alaudet/raspi-sump/blob/master/docs/install.md I was expecting to see a new /home/<removed>/raspi-sump ceated.  However I didn't see one get created.  I did notice upon uninstalling that pip seems to think raspi-sump is now located elsewhere.  See below:

```
$ sudo pip3 uninstall raspisump
Found existing installation: raspisump 1.9.3
Uninstalling raspisump-1.9.3:
  Would remove:
    /usr/local/bin/emailtest
    /usr/local/bin/rsump.py
    /usr/local/bin/rsumpchart.py
    /usr/local/bin/rsumpmonitor.py
    /usr/local/bin/rsumpsupport
    /usr/local/bin/rsumpwebchart.py
    /usr/local/lib/python3.9/dist-packages/home/<removed>/.config/systemd/user/raspisump.service
    /usr/local/lib/python3.9/dist-packages/home/<removed>/.config/systemd/user/rsumpwebchart.service
    /usr/local/lib/python3.9/dist-packages/home/<removed>/.config/systemd/user/rsumpwebchart.timer
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/VERSION
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/charts/README.md
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/cron/README.md
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/cron/picrontab
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/csv/README.md
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/docs/README.md
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/docs/install.md
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/logs/README.md
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/sample_config/raspisump.conf
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/web/css/inc/VERSION
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/web/css/includes.js
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/web/css/index.html
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/web/css/raspi.css
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/web/images/logo.png
    /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump/web/index.html
    /usr/local/lib/python3.9/dist-packages/raspisump-1.9.3.dist-info/*
    /usr/local/lib/python3.9/dist-packages/raspisump/*
```

If this is working as you expect, is /home/<removed>/raspi-sump actually a symlink to /usr/local/lib/python3.9/dist-packages/home/<removed>/raspi-sump ?  If so, should I just make the symlink manually?  Or is there something larger going on here?

alaudet, 2023-08-12:
This is normal, you dont have to rename anything.   

/home/user/raspi-sump contains all of the user environment stuff (conf files, charts, csv, logs etc etc).  /usr/local/bin and /usr/local/lib.... is where executables and llibraries are installed on the system.

No symlinks are needed, after you install the executables are in your path so can access rsump.py, rsumpwebchart.py etc etc from anywhere.

By renaming the /home/user/raspi-sump folder you would have created a new one on install and now have to recreate your raspisump.conf file.  If you leave it, it would have saved it.

Is it working now?

alaudet, 2023-08-15:
I don't know, something seems off to me, maybe corrupted on your install.  Also, again, do not run rsump.py with sudo.  just run rsump.py directly.  You install with sudo, you run the program as a regular user, not with admin privilidedges

The following has always worked for me;

**Remove**
- cd ~    (to make sure you are in the home user folder)
- sudo pip3 uninstall raspisump
- rm raspi-sump -r -f

**Install fresh**  

- sudo pip3 install --no-binary :all: raspisump
- cd raspi-sump
- mv sample_config/raspisump.conf  .         (note that the period specifies the current folder)
- nano raspisump.conf       (enter your variables)

Now follow the instructions for systemd in the doc.

fitig, 2023-08-18:
I have most everything working now.  It looks like there were some items that didn't get added to the raspisump.conf file as part of the upgrade that prevented the service from starting.  That's what was causing the error above.  I added those by looking at the default raspisump.conf file.  So now the raspisump service and rsumpwebchart timer start successfully.  However, I'm not able to enable either one of them.  When I try to enable them here are the errors I get:

```
$ systemctl --user enable raspisump.service
Failed to enable unit: Unit file /home/user/.config/systemd/user/multi-user.target.wants/raspisump.service does not exist.
$ systemctl --user enable rsumpwebchart.timer
Failed to enable unit: Unit file /home/user/.config/systemd/user/timers.target.wants/rsumpwebchart.timer does not exist.
```

I have tried creating either of the directories that it mentions and copying the .service and .timer files from their default location into the respective directories and that then causes an access denied error.  Any idea what might be going on with not being able to enable the service and the timer?

alaudet, 2023-08-20:
Now that it is working perhaps rerun the `rsumpsupport` utility and send me the output.

fitig, 2023-08-22:
Unfortunately it still throws this:
$ rsumpsupport
```
Traceback (most recent call last):
  File "/usr/local/bin/rsumpsupport", line 71, in <module>
    raspi_sump_version = get_raspi_sump_version(actual_user)
  File "/usr/local/bin/rsumpsupport", line 28, in get_raspi_sump_version
    with open(f"/home/{actual_user}/raspi-sump/VERSION", "r") as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/user/raspi-sump/VERSION'
```

Any chance you can give me a properly formatted VERSION file?  The installer didn't create that for me.

alaudet, 2023-08-22:
Did you create a user folder?  User should show your actual username like pi.   Something is off with your install

fitig, 2023-08-22:
Isn't the user folder just the home directory of the user who I run the install command as?  If yes, then its my home directory so it was already there.  Inside of that is where my raspi-sump directory is.  Inside of that directory (i.e. /home/myuserid/raspi-sump) is where I think the rsumpsupport script expects to find a VERSION file.  That's why I was asking for a VERSION file.  I ended up just grabbing the one from the root of the repo and put in there.  Output is attached.

Note that the service and timer were not started because I recently rebooted my pi from doing updates.  I started the service shortly before I ran the rsumpsupport script.  This is why I'm trying to get the service enabled so that it'll automatically start when my pi boots up.

In looking at it, I'm not seeing anything obvious that tells me why I'm getting the error upon enabling the service and the timer.  Let me know if anything sticks out to you.  Thanks again for all the help!

[support-20230822_094936.txt](https://github.com/alaudet/raspi-sump/files/12409461/support-20230822_094936.txt)

alaudet, 2023-08-24:
I have never seen a user's home .config symlinked to a system folder.  It must have been that way before your upgrade, I don't mess with that in my install script.
The reason I use the systemd user context for raspisump is that raspisump is run as the user.  There is a lot of legacy decisions I made with config files and the whole environment in the user folder.  I want raspisump running as the user and not the system itself.

I am planning a redesign of raspisump in the future as python is getting a lot more picky in how applications are installed.  The latest debian bookworm release does not even work with the current install instructions at all.  It doesnt allow install to the system Python without further install command hacks.    So I need to rethink my approach.  

Mostly this application is run on a pi that is dedicated to one task and this approach makes it a bit easier as all files are under user home.

I am evaluating how I want to setup the whole environment in the future that will make it easier to deploy and easy for users to configure.   Maybe within a virtual environment or within a docker container or something like that.  I can also look at creating deb files so that I could submit them to the raspios package maintainers to install with apt, but thats another rabbit hole.

This issue seems like a one off situation and gladly you solved it. So I am going to close the issue.

thanks

closed 2023-08-24 by alaudet