alaudet/raspi-sump #83: Can connect my sensor hcsr04 to the raspberry 4b

Bizz101, 2024-08-12:
I read a lot on your GitHub … change a bunch of sensor , wires , even change voltage divider (1k and 2k) (330ohms and 470ohms) . 

I always received a error log 
Signal not received. Possible cable or sensor problem

also did the command line 
Hcsr04.py -t 17 -e 27 

echo line not received any signal 

I try to uninstall raspisump , no succes

I check in groups just to be sure GPIO are in my user … and yes it is 

I’m not a advanced user . But I build a craft beer pi without any issue and I really need your project to monitor a closed sewage tank 

Tks for your help

Bizz101, 2024-08-21:
I work several hours and here what I found …

bought a bunch of hcsr04 sensor , none of them work 

also with the jst-04 sensor but led working ( wow) 

got proper voltage a vcc pin on sensor 

voltage divider is correct , I checked with power supply , got 3 volt at the end ( 1k ohm and 2 k ohm) 

trigger will not send signal , check with GPIO test, everything is right , code also , GPIO 17 output , GPIO 27 input 

send a 3 volt signal to echo pin  when I make a test with hcsr04.py , got a reading … it look like a reading … well better than no echo 

any info will be appreciated 

tks 

OS 11 (legacy) 64 bit just in case it could make a issue

alaudet, 2024-08-23:
Hi I havent tested with 64bit but it should not matter.  I will have to test it in the next few days.  Have you tried with 32 bit RaspiOS?

Bizz101, 2024-08-24:
Ill try it this week and I’ll give you a report if it’s the 64bits OS create the issue .
Bastien Bisaillon

Bizz101, 2024-08-25:
ok, now using a code i can use my sensor and it work ... problem is when i make change raspisump in config file of raspi sump the sensor is not working! hcsr04.py failed also ? here the report of raspisump

 RaspiSump Service
     Loaded: loaded (/home/pi/.config/systemd/user/raspisump.service; disabled; vendor preset: enabled)
     Active: failed (Result: start-limit-hit) since Sun 2024-08-25 17:14:22 EDT; 6s ago
    Process: 1971 ExecStart=/usr/local/bin/rsump.py (code=exited, status=0/SUCCESS)
   Main PID: 1971 (code=exited, status=0/SUCCESS)
        CPU: 215ms

Aug 25 17:14:22 raspberrypi systemd[592]: raspisump.service: Scheduled restart job, restart counter is at 5.
Aug 25 17:14:22 raspberrypi systemd[592]: Stopped RaspiSump Service.
Aug 25 17:14:22 raspberrypi systemd[592]: raspisump.service: Start request repeated too quickly.
Aug 25 17:14:22 raspberrypi systemd[592]: raspisump.service: Failed with result 'start-limit-hit'.
Aug 25 17:14:22 raspberrypi systemd[592]: Failed to start RaspiSump Service.

tks for your help

Bizz101, 2024-08-28:
Here the code who works for my sensor… if I’m correct it’s look it’s trigger at high first and then the low pulse after time sleep.Yours start with low impulse , then high , time sleep and low . Don’t know if it’s the issue why it’s not working …  I’ll make some test and get you some news if I change the code . 

Here the code that work with hcsr04 on OS11 64bits

Libraries
import RPi.GPIO as GPIO
import time
 
#GPIO Mode (BOARD / BCM)
GPIO.setmode(GPIO.BCM)
 
#set GPIO Pins
GPIO_TRIGGER = 18
GPIO_ECHO = 24
 
#set GPIO direction (IN / OUT)
GPIO.setup(GPIO_TRIGGER, GPIO.OUT)
GPIO.setup(GPIO_ECHO, GPIO.IN)
 
def distance():
    # set Trigger to HIGH
    GPIO.output(GPIO_TRIGGER, True)
 
    # set Trigger after 0.01ms to LOW
    time.sleep(0.00001)
    GPIO.output(GPIO_TRIGGER, False)
 
    StartTime = time.time()
    StopTime = time.time()
 
    # save StartTime
    while GPIO.input(GPIO_ECHO) == 0:
        StartTime = time.time()
 
    # save time of arrival
    while GPIO.input(GPIO_ECHO) == 1:
        StopTime = time.time()
 
    # time difference between start and arrival
    TimeElapsed = StopTime - StartTime
    # multiply with the sonic speed (34300 cm/s)
    # and divide by 2, because there and back
    distance = (TimeElapsed * 34300) / 2
 
    return distance
 
if __name__ == '__main__':
    try:
        while True:
            dist = distance()
            print ("Measured Distance = %.1f cm" % dist)
            time.sleep(1)
 
        # Reset by pressing CTRL + C
    except KeyboardInterrupt:
        print("Measurement stopped by User")
        GPIO.cleanup()

alaudet, 2025-01-25:
Hi I am not sure what to do with this issue.  I have tested over and over and cannot replicate this issue.  I will mark it closed but if you need help please feel free to reopen it.

closed 2025-01-25 by alaudet